pingumacpenguin/STM32-O-Scope #1: ADC output order

DavidPilling, 2016-02-04:
I believe that in serialSamples() ADC1 and ADC2 are being output in the wrong order.

pingumacpenguin, 2016-02-05:
So... do I close this issue, or is there a problem that needs to be addressed?

DavidPilling, 2016-02-05:
I believe there is a problem. The serialSamples() code outputs ADC1 then ADC2 but ADC2 is read first. When I look at plotted samples the lines jump back on themselves. Quoting the reference manual

"The bytes are coded in memory in Little Endian format. The lowest numbered byte in a word
is considered the word’s least significant byte and the highest numbered byte the most
significant."

and 

"containing the ADC2 converted data in the upper halfword and the
ADC1 converted data in the lower halfword"

and

" ADC2 starts immediately and  ADC1 starts after a delay of 7 ADC clock cycles"

pingumacpenguin, 2016-02-06:
Well spotted, its been a while since I looked at that code, can you suggest the fix to, save me trawling through the code? Presumably we need to loop through the sample array two elements at a time, and output the values 2nd then 1st rather than the current method of simply looping through the array one element at a time. 

Presumably this will also affect the plotting of values on the display too.

DavidPilling, 2016-02-06:
This is what I did:

 for (int16_t j = 0; j < maxSamples   ; j++ )
 {
  int i;
  if(j & 1) i=j-1;
  else      i=j+1;

  pr("%f\t%d",(samplingTime / (maxSamples))*i,dataPoints[i]);
 
 }

Obviously using my own pr() serial print function. Just replace j with i after the first three lines.

Yes as far as I can tell display plotting is affected too.

Many thanks for the code and the forum postings - I've found it all very useful.

Feel free to close this issue.

pingumacpenguin, 2016-03-28:
> Feel free to close this issue.
Will do.

closed 2016-03-28 by pingumacpenguin