pingumacpenguin/STM32-O-Scope #4: Ram over flow on stm32f103c86 with 17k ram

KiranKanchi, 2017-12-10:
Hello,
          Thank you so much for sharing your code ,
i tried running your software in my setup, however i get an error 

c:/users/kiran kanchi/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld.exe: C:\Users\KIRANK~1\AppData\Local\Temp\arduino_build_396000/STM32-O-Scope.ino.elf section `.bss' will not fit in region `ram'

c:/users/kiran kanchi/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld.exe: region `ram' overflowed by 304 bytes

collect2.exe: error: ld returned 1 exit status
using arduino 1.8.5
any work around ?
How much was your RAM ?
[image]

pingumacpenguin, 2017-12-22:
Try decreasing the size of the sample buffer.

victorpv, 2017-12-22:
Quick question, why do you say 17K of RAM? The mcu should have 20KB. If you are using a board selection that limits the RAM to 17KB, something is wrong and old. That was a limit set in the old libmaple core, but was removed a long time ago since all 20KB are available after the bootloader jumps to a sketch.

KiranKanchi, 2017-12-22:
I tried 20k variant . Compilation fails. 
Perhaps i am using old maple core

pingumacpenguin, 2017-12-29:
> Perhaps i am using old maple core
If you can verify this was the cause, I will close this issue. 
If I haven't heard from you within 7 days, I'll close as "problem not confirmed", but if you believe this issue is still valid, provide full details and I will investigate further.