pingumacpenguin/STM32-O-Scope #7: Cannot build

grego1981, 2018-09-28:
Hi all,

I'm using STM32F103C816 and I'm trying to build via Arduino IDE 1.8.6 and I get the errors below, I have installed the latest stm libraries and hardware and have uploaded the blink test with success...... How can I remedy this errors???

Build options changed, rebuilding all
D:\data\gdrive\BenchPSU\oscilloscope\STM32-O-Scope\STM32-O-Scope.ino: In function 'void showTime()':

STM32-O-Scope:598:16: error: 'hour' was not declared in this scope

     if (hour(tt) < 10) {

                ^

STM32-O-Scope:601:22: error: 'hour' was not declared in this scope

     TFT.print(hour(tt));

                      ^

STM32-O-Scope:603:18: error: 'minute' was not declared in this scope

     if (minute(tt) < 10) {

                  ^

STM32-O-Scope:606:24: error: 'minute' was not declared in this scope

     TFT.print(minute(tt));

                        ^

STM32-O-Scope:608:18: error: 'second' was not declared in this scope

     if (second(tt) < 10) {

                  ^

STM32-O-Scope:611:24: error: 'second' was not declared in this scope

     TFT.print(second(tt));

                        ^

STM32-O-Scope:613:21: error: 'day' was not declared in this scope

     TFT.print(day(tt));

                     ^

STM32-O-Scope:615:23: error: 'month' was not declared in this scope

     TFT.print(month(tt));

                       ^

STM32-O-Scope:617:22: error: 'year' was not declared in this scope

     TFT.print(year(tt));

                      ^

D:\data\gdrive\BenchPSU\oscilloscope\STM32-O-Scope\STM32-O-Scope.ino: In function 'void setCurrentTime()':

STM32-O-Scope:867:26: error: 'setTime' was not declared in this scope

   setTime(thisArg.toInt());

                          ^

STM32-O-Scope:868:19: error: 'now' was not declared in this scope

   time_t tt = now();

                   ^

D:\data\gdrive\BenchPSU\oscilloscope\STM32-O-Scope\STM32-O-Scope.ino: In function 'void serialCurrentTime()':

STM32-O-Scope:875:14: error: 'hour' was not declared in this scope

   if (hour(tt) < 10) {

              ^

STM32-O-Scope:878:29: error: 'hour' was not declared in this scope

   serial_debug.print(hour(tt));

                             ^

STM32-O-Scope:880:16: error: 'minute' was not declared in this scope

   if (minute(tt) < 10) {

                ^

STM32-O-Scope:883:31: error: 'minute' was not declared in this scope

   serial_debug.print(minute(tt));

                               ^

STM32-O-Scope:885:16: error: 'second' was not declared in this scope

   if (second(tt) < 10) {

                ^

STM32-O-Scope:888:31: error: 'second' was not declared in this scope

   serial_debug.print(second(tt));

                               ^

STM32-O-Scope:890:28: error: 'day' was not declared in this scope

   serial_debug.print(day(tt));

                            ^

STM32-O-Scope:892:30: error: 'month' was not declared in this scope

   serial_debug.print(month(tt));

                              ^

STM32-O-Scope:894:29: error: 'year' was not declared in this scope

   serial_debug.print(year(tt));

                             ^

exit status 1
'hour' was not declared in this scope

zoomx, 2018-09-28:
Not sure but change this
#include "Time.h" 
in this
#include "TimeLib.h"

grego1981, 2018-10-01:
Yes renaming the Time library, did the trick!
Thank you!

closed 2018-10-30 by pingumacpenguin