pingumacpenguin/STM32-O-Scope #13: O-Scope does't draw a graph

Ledrunning, 2020-04-13:
I used newest [STM32DUINO ](url)Core and downloaded your sketch compiled and uploaded it to the device via UART. Then I started the device. The grid and coordinate lines were displayed on the screen. Also on the screen were displayed inscriptions below 0.0 uS/Sample etc... But any noise or pulse signal from PB1 on my Probe.
So I've tryed to log my steps in Usart
`void takeSamples ()
{
  // This loop uses dual interleaved mode to get the best performance out of the ADCs
  //
  Serial.println("Init DMA");
  dma_init(DMA1);
  dma_attach_interrupt(DMA1, DMA_CH1, DMA1_CH1_Event);
  Serial.println("Enable DMA for ADC");
  adc_dma_enable(ADC1);
  dma_setup_transfer(DMA1, DMA_CH1, &ADC1->regs->DR, DMA_SIZE_32BITS,
                     dataPoints32, DMA_SIZE_32BITS, (DMA_MINC_MODE | DMA_TRNS_CMPLT));// Receive buffer DMA
  Serial.println("Set DMA transfer");
  Serial.println(maxSamples / 2);
  dma_set_num_transfers(DMA1, DMA_CH1, maxSamples / 2);
  dma1_ch1_Active = 1;
  //  regs->CR2 |= ADC_CR2_SWSTART; //moved to setADC
   Serial.println("Enable the channel and start the transfer");
  dma_enable(DMA1, DMA_CH1); // Enable the channel and start the transfer.
  //adc_calibrate(ADC1);
  //adc_calibrate(ADC2);
  samplingTime = micros();
  Serial.println(samplingTime);
  while (dma1_ch1_Active); // SOME BUG OR WHAT.... PROGRAM STOP HERE!!!
  samplingTime = (micros() - samplingTime);
  Serial.println("Disable DMA");
  dma_disable(DMA1, DMA_CH1); //End of trasfer, disable DMA and Continuous mode.
  // regs->CR2 &= ~ADC_CR2_CONT;

}`

I think programm can't exit from while loop, maybe dma event does not work?
Any suggestion?

pingumacpenguin, 2020-04-27:
Its been a long time since I looked at this code, but I suggest you debug by the time honoured process of sticking a 

Serial.println("Whatever.."); 

Statement in the code, and keep moving it down until it stops showing the "Whatever" on your serial console. 

The issue lies with the previous statement. 

If that calls a function, move the serial print into that function.. rinse, repeat.. until you find where the code is stopping. 

My gut feeling from your comment above is that the crude triggering algorithm is not working. 

This could indeed be DMA related. 

The code was originally run on Roger's core, so I have no idea if it works with the official STM core. 
Keep me posted if you make any progress. Better still, if you figure it out, fire me a pull request and I'll test and merge your change(s). 

If you are getting nowhere, let me know and I'll dig out my STM stuff and take a look. I have been meaning to get back in to the stmduino world, but currently I'm on a bit of a watch repair binge (and trying to keep on top of the gardening, since it is spring here, and all of the weeds have gone into overdrive).

Ledrunning, 2020-04-27:
Yes, I tried to blink and log wrote to serial. Program stops on while loop when you used DMA flag and static function. Then I tested DMA exaples from Roger's core and got the same problem. So I created Issue [there](https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/769).